cwalter-at/freemodbus #2: When Write MULTIPLE_COILS

ixuhan, 2019-05-07:
if i send : 01 0F 00 00 00 03 01 0f CF 53
i means wirte three coils to bit 1,but it returns four bits to bit 1,
if we should just get three bits in 0x0f ?

JoelStienlet, 2019-05-14:
Hello,
Do you use xMBUtilSetBits() of mbutils.c to set the bits when in the eMBRegCoilsCB() callback?

ixuhan, 2019-05-14:
of course. i think eMBRegCoilsCB():
`case MB_REG_WRITE:
                while( iNCoils > 0 )
                {
    			    UCHAR ucCoilsNums = iNCoils > 8 ? 8 : iNCoils;
    			    UCHAR ucByte = *pucRegBuffer++ & ( UCHAR )( ( 1 << ( UCHAR ) ucCoilsNums ) - 1 );
                    xMBUtilSetBits( pHTD_HoldRegSelfModelValue, usBitOffset,
                                    ( unsigned char )( iNCoils > 8 ? 8 : iNCoils ),
                                    ucByte );
                    iNCoils -= 8;
                    usBitOffset += 8;
                }`

JoelStienlet, 2019-05-14:
I think this problem could be corrected by inserting:
`usValue &= ( USHORT )( ( 1 << ( USHORT ) ucNBits ) - 1 );`
after `usValue <<= usNPreBits;` in xMBUtilSetBits().

ixuhan, 2019-05-14:
ok,it works well,thks!

closed 2019-05-14 by ixuhan