cwalter-at/freemodbus #14: In mbfuncholding.c, line 185 is there a bug?

rhino949, 2019-11-06:
freemodbus/modbus/functions/mbfuncholding.c;
line 185,  usRegCount = ( USHORT )( pucFrame[MB_PDU_FUNC_READ_REGCNT_OFF + 1] );
like should be: usRegCount |= ( USHORT )( pucFrame[MB_PDU_FUNC_READ_REGCNT_OFF + 1] );

JoelStienlet, 2019-11-06:
Indeed, I agree this looks like a bug: the 8 upper bits get trashed. The function will sometimes succeed where it should have returned "eStatus = MB_EX_ILLEGAL_DATA_VALUE;"

JoelStienlet, 2019-11-08:
@cwalter-at can you please make this very small change, just add the "|" at L185?

ntfreak, 2020-01-15:
see https://github.com/cwalter-at/freemodbus/pull/4/files

JoelStienlet, 2020-01-15:
So, given the good explanations of karlp on this link, perhaps that first line (L184) could indeed be dropped? The compiler probably drops it anyway (when optimizations are enabled), but that would make the code easier to understand IMO.

rhino949, 2020-01-20:
I got it.

closed 2020-01-20 by rhino949