cwalter-at/freemodbus #16: queston about eMBFuncReadInputRegister

glct12fsl, 2019-12-09:
at line 71 of mbfuncinput.c there is an increment  of usRegAddress variable
before calling at line 96 the function eMBRegInputCB().
i think, the function eMBRegInputCB() will read the next register address of the desired.

George

yannickasselin, 2020-03-10:
I noticed the same thing. Every read/write functions have this increment of usRegAddress. So when testing modbus TCP using Codesys I always end up reading/writing the wrong registers (Offset by 1 more than desired). I know some modbus device registers start at 1 others starts at 0. This may be the reason why but I don't quite understand. Can anybody give an explanation?

karlp, 2020-05-05:
freemodbus does this +1 to just hard code all values to be "register numbers" ie, 1 based.  If you want to work with "register addresses" ie, 0 based, you must subtract one in the beginning of your user callbacks.  Should this be configurable? Maybe. Would it perhaps have been simpler to not do the +1? Maybe, but that's how it is.

closed 2025-05-30 by pvyleta