cwalter-at/freemodbus #21: mbfuncdisc.c line 54 Is this string correct?

loki-amorf, 2020-10-23:
https://github.com/cwalter-at/freemodbus/blob/f16701094ca64df3a0366dde1c186a55976d61e4/modbus/functions/mbfuncdisc.c#L54
Why do we check if read coils were enabled?
Why don't we check if read discrete inputs were enabled?
I gues it should be something like #if MB_FUNC_READ_DISCRETE_INPUTS_ENABLED > 0

CyrilJiang, 2022-03-12:
Yes，I think so.

pvyleta, 2025-06-30:
Hi @loki-amorf @CyrilJiang , this project sees a very little activity which I am trying to change, but both my time and my capabilities to test the code are limited. If you create a PR and you are able to present some basic information as to how this was tested and explain the nature of the changes, I would happily merge the fix.