cwalter-at/freemodbus #23: usRcvBufferPos reset only in STATE_RX_IDLE in mbrtu.c

tortuoise, 2020-11-18:
Firstly, thanks to the authors for this nice concise modbus slave ported to multiple devices. I tested it on a MSP430F5529 so made a few changes to the provided MSP430F169 port. I got it working but only after addressing a couple of issues which I think should impact all devices. 
The issue I had to fix had to do with the current position of the buffer use to store incoming bytes (usRcvBufferPos) which only gets reset in STATE_RX_IDLE. If for whatever reason a slave doesn't respond to a request then the start position of the buffer doesn't get reset and is in an unknown state when the next request comes in. Setting it back to zero at the ***end*** of receiving the current frame seems to fix the problem though I'm not sure whether it introduces other problems. 

`eMBErrorCode eMBRTUReceive( UCHAR * pucRcvAddress, UCHAR ** pucFrame, USHORT * pusLength )  
{
  ...
   usRcvBufferPos = 0;
   return eStatus;
}
`

There's probably a better solution to this. Anybody else encounter this issue?

CyrilJiang, 2022-03-12:
the timer and if no character is received within t3.5 we change to STATE_RX_IDLE. This makes sure that we delay startup of the  modbus protocol stack until the bus is free.

pvyleta, 2025-05-30:
@CyrilJiang @tortuoise what is the conclusion on this one? is this a real issue to be solved?

closed 2025-07-04 by pvyleta