cwalter-at/freemodbus #25: RTU only responds to 35 messages

Ezraneut, 2021-01-20:
I've ported the code to be used on a atXMega32D4 and got it working with several Master applications.

The common thing that is not working is that after 35 messages, the USART completely stops receiving any data into the DATA register.

Did anyone ever encounter anything like this? Do I need to clear the DATA register somewhere again?

matthiesenj, 2021-01-22:
I've used freemodbus RTU for over 3 years on ATMega128 and never encountered any problems - you probably have a bug in your porting code.

Ezraneut, 2021-01-22:
Thanks for your reply.

I've been sending raw values over the serial bus into the firmware and noticed that the communication blocks mid frame when sending >2 bytes in 1 USART frame. When sending 1 value in a frame, it runs continuously. Indeed this has to be an issue in how I ported either the timer or the USART configurations. 

I will dig deeper into it and hope to report good news :)

Ezraneut, 2021-01-22:
Alright, I have found my own bug.

I am using the asf libraries of Atmel, and used the asf functions to turn the timer on and off using the function tc_enable() and tc_disable(). In the functions a counter for sleep modes is kept, and raised 1 every time the clock is turned on, and subtracted 1 when turning off.

I called this function every time the counter was started, and this is multiple times within 1 frame. So everytime a frame with multiple bytes was received, the sleep counter was raised multiple times, but only subtracted once.

Within the sleep counter, the following code made my code lock when the sleepmgr_locks[mode] was above 255.
```
if(sleepmgr_locks[mode] >= 0xff) {
		while (true) {
			// Warning: maximum value of sleepmgr_locks buffer is no more than 255.
			// Check APP or change the data type to uint16_t.
		}
	}
```

closed 2021-01-22 by Ezraneut