cwalter-at/freemodbus #37: freemodbus design question

matthiesenj, 2023-01-31:
I'm wondering if someone can tell me what the rationale is behind the EV_EXECUTE event in eMBPoll?

It is only invoked if the handler for EV_FRAME_RECEIVED posts EV_EXECUTE, but why isn't the code for EV_EXECUTE just included in the handler for EV_FRAME_RECEIVED? It would make the setup simpler and use less resources in a scheduled environment (e.g. FreeRTOS) because 1) there would be less (unnecessary) kernel calls, and 2) in many cases you would have ISR to task events only (i.e. no task to same task event). This is assuming you implement the events using a kernel queue mechanism.

Some of the above goes for the EV_FRAME_SENT event also, which doesn't do anything.

pvyleta, 2025-05-30:
Hi @matthiesenj, probably the only person who could answer the question is you, as nobody else seems to know. Would you mind sharing your findings?

pvyleta, 2025-10-19:
Hi @matthiesenj, if you feel this is worth changing for everyone, feel free to submit a PR.