cwalter-at/freemodbus #38: BARE demo portevent.c variables have a lack of volatile qualifier. No event received from a bus with optimization enabled.

Nikolay-Po, 2023-04-24:
Demo portevent.c variables are optimized-out by current compilers. No events received from a bus.
Here, demo/BARE/port/portevent.c:

```
/* ----------------------- Variables ----------------------------------------*/
static eMBEventType eQueuedEvent;
static BOOL     xEventInQueue;

```
Both variables MUST be declared with volatile qualifier:

```
/* ----------------------- Variables ----------------------------------------*/
static volatile eMBEventType eQueuedEvent;
static volatile BOOL     xEventInQueue;

```
This is because both variables are changing from an interrupts, unexpectedly for the compiler.
When I compiled my RTU port using BARE demo template, I discovered it works without an optimization and doesn't with high level.
The single failure point was portevent.c with variable declaration. It is obvious that these variables need to be volatile to be processed correctly in MODBUS poll function.

pvyleta, 2025-05-30:
Hi @Nikolay-Po , this project sees a very little activity which I am trying to change, but both my time and my capabilities to test the code are limited. If you create a PR and you are able to present some basic information as to how this was tested and explain the nature of the changes, I would happily merge the fix.

Nikolay-Po, 2025-05-31:
Done. From code flow point of view, the addition of **volatile** qualifier to static variables is safe and can't break anything except of small increase of execution time and code size. With the **volatile** code word the compiler had to assume the variable can change suddenly, out of compiler scope and out of compiler control. So, the optimization actions of a compiler for such variable are strictly limited. The compiler must really access a memory where the variable is stored every time it appears in the code. So, the change of a variable in an interrupt will not be ignored.
In contrast, without of **volatile** qualifier, with high optimization demand, a compiler will ignore the variable at all because it doesn't change during of execution of current function.
I tested only the code from /demo/BARE part of repository in three different projects with ModbusRTU servers. But having an experience with Microchip PIC, AVR, ARM and RISC-V architectures in bare metal projects I sure this modification is correct for all bare metal versions as requested in PR #50.

closed 2025-06-04 by pvyleta