cwalter-at/freemodbus #54: mbconfig options

PaulFirs, 2025-06-15:
### **Problem**  
Currently, Modbus protocol flags (`MB_ASCII_ENABLED`, `MB_RTU_ENABLED`, `MB_TCP_ENABLED`) are defined without `#ifndef` guards. This may lead to:  
- **Redefinition errors** if the header is included multiple times.  
- **No flexibility** to override defaults via compiler flags.  

### **Proposed Solution**  
Add conditional compilation:  
```c
#ifndef MB_ASCII_ENABLED  
#define MB_ASCII_ENABLED (1)  
#endif  

#ifndef MB_RTU_ENABLED  
#define MB_RTU_ENABLED (1)  
#endif  

#ifndef MB_TCP_ENABLED  
#define MB_TCP_ENABLED (0)  
#endif

PaulFirs, 2025-06-15:
Created [PR](https://github.com/cwalter-at/freemodbus/pull/55)

pvyleta, 2025-06-22:
Thanks for the suggestion. I see that not all defines in the given header have the ifdef guards. Would you believe it would make sense to add the ifdef guards to all defines? If so, would you be willing to make that improvement to keep the code consistent? Also - please provide teh description in this issue in the PR itself, squash the commits, and I will be happy to include your suggestion. once again, thanks a lot.

PaulFirs, 2025-06-23:
@pvyleta 
Created PR https://github.com/cwalter-at/freemodbus/pull/57
Please let me know if any adjustments are needed. Happy to refine further!
Thanks for your guidance.